the-modem-distro/pinephone_modem_sdk #247: 0.8.0: if phone is in suspend, incoming sms are always reported as stuck messages

bircoph, 2025-01-14:
Hello!

After upgrading modem 0.7.2 -> 0.8.0 and ADSP 30.006 -> 30.007 I encounter the following issue:
If the phone (pinephone v1.2) is in the sleep state (e.g. configured to suspend after inactivity on battery power) and an incoming SMS is received, the phone wokes up and the modem always sends a message from internal number about a stuck message.

A sender is reported correctly, but the text first part of the message is always omitted, but if the message is long enough to be multi-part, second and later parts (if any) are displayed correctly.

I downgraded ADSP to 30.006, rebooted phone, but the problem persists.
Then I tried AT+WIPESMS and phone reboot, but still no effect.

Is any way to disable or modify stuck message handling? Something is wrong with it.

bircoph, 2025-01-14:
> Is any way to disable or modify stuck message handling? Something is wrong with it.

Yes, there is a way:
```
disable message recovery
```
via sms interface.

I missed this originally, because this command is not present in the https://github.com/the-modem-distro/pinephone_modem_sdk/blob/scarthgap/docs/SMS_INTERFACE.md However, it is available via `help` command.

This fixes my issue, but it is better to fix message recovery instead.

sammyj85, 2025-01-30:
@bircoph does disabling the stuck message handling mean the message is lost? Or can you describe how it behaves with this change?