the-modem-distro/pinephone_modem_sdk #248: sms issues

giorez, 2025-01-26:
I'm actually using PPPee, 0.8.0 modemdistro, ADSP EG25GGBR07A08M2G_30.202.30.202.

I followed the troubleshoting to solve double sms issue. Unfortunatelly I receive 7/8 copies of the same sms.
Plus I can't any response when sendind sms to +223344556677. Any help to investigate? thanks

mikeshenson, 2025-02-07:
What Operating System are you using? 
Are you using mmsd-tng or ofono?

giorez, 2025-02-08:
i m using arch linux with phosh. how do i know if those are based on mmsd-tng or ofono? thanks

mikeshenson, 2025-02-08:
Arch with phosh should be using mmsd-tng. 
$sudo pacman -Ss mmsd-tng
this should return "danctnix/mmsd-tng 2.6.2-2 [installed]"

Have you updated the whole OS lately? I have a fully updated systems as of 2 weeks ago.

Device: PinePhonePro (Daily driver)
OS: DanctNIX Arch with Phosh
Carrier: USA - Patriot Mobile (T-Mobile SIM)
EG25 Firmware: Biktorgj 0.8.0
ADSP Version: 01.003

giorez, 2025-02-08:
danctnix/mmsd-tng 2.6.2-2 [installed]
just updated while ago

giorez, 2025-02-08:
since the original install

mikeshenson, 2025-02-08:
Here are some screen shots of all the setting I have in chatty.

What carrier are you using?

<p align="center">
  [image]
  [image]
  [image]
</p>

giorez, 2025-02-08:
@mikeshenson I activated "clear stuck sms". I recheck if it worked out. thanks

giorez, 2025-02-20:
It seems the problem is still there...

LinLo, 2025-02-24:
Hi !

I have quite same issue since months, I don't know exactly when this has started.. Sometimes I'm receiving the same SMS multiple times during multiple days. Sometimes WIPESMS is helping, sometimes not...
 
I tried many modem distro versions and reverted back to different stock firmware versions but same issue.

I don't know if it is a firmware issue or a software issue.

Have you tried the 30.204.30.204 modem firmware version ? Is it better ?

mikeshenson, 2025-02-24:
@LinLo and @giorez what mobile carrier are you using? There is a list of carriers and the best ADSP to use.

https://github.com/the-modem-distro/pinephone_modem_sdk/blob/scarthgap/docs/ADSP-CARRIERS.md

giorez, 2025-02-25:
i ve tryed to use 30.204, it doesn't solve the issue. i'm using spusu (virtual operatore on wind3 operator)

mikeshenson, 2025-02-25:
@giorez, Thank you for sharing the operator. I would look into any issues with "mmsd-tng"[1] package. 
run this command and see if mmsd-tng is giving you errors or warnings. 
`journalctl | grep mmsdtng`

[1] https://gitlab.com/kop316/mmsd

giorez, 2025-04-25:
sorry for late reply, I didnt use the phone in last months. here is the mmsdtng response.

`_STATE_READY
Apr 24 13:08:59 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:mmsd_mm_state() Setting
 Bearer Handler
Apr 24 13:08:59 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service.c:mms_service_set_bearer_handler()
 service 0xaaab09610e80 handler 0xaaaadd78afc0
Apr 24 13:08:59 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:modem_state_changed_cb(
) State Change: Old State: 9 New State: 8, Reason: 1
Apr 24 13:08:59 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:mmsd_mm_get_modem_state
() MM_MODEM_GOOD_STATE: 8
Apr 24 13:08:59 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:mmsd_mm_state() MMSD_MM
_STATE_READY
Apr 24 13:08:59 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:mmsd_mm_state() Setting
 Bearer Handler
Apr 24 13:08:59 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service.c:mms_service_set_bearer_handler()
 service 0xaaab09610e80 handler 0xaaaadd78afc0
Apr 24 19:40:43 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:modem_state_changed_cb(
) State Change: Old State: 8 New State: 10, Reason: 1
Apr 24 19:40:43 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:mmsd_mm_get_modem_state
() MM_MODEM_GOOD_STATE: 10
Apr 24 19:40:43 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:mmsd_mm_state() MMSD_MM
_STATE_READY
Apr 24 19:40:43 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:mmsd_mm_state() Setting
 Bearer Handler
Apr 24 19:40:43 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service.c:mms_service_set_bearer_handler()
 service 0xaaab09610e80 handler 0xaaaadd78afc0
Apr 24 19:40:44 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:modem_state_changed_cb(
) State Change: Old State: 10 New State: 11, Reason: 1
Apr 24 19:40:44 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:mmsd_mm_get_modem_state
() MM_MODEM_GOOD_STATE: 11
Apr 24 19:40:44 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:mmsd_mm_state() MMSD_MM
_STATE_READY
Apr 24 19:40:44 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:mmsd_mm_state() Setting
 Bearer Handler
Apr 24 19:40:44 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service.c:mms_service_set_bearer_handler()
 service 0xaaab09610e80 handler 0xaaaadd78afc0
Apr 24 19:40:44 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:process_mms_process_mes
sage_queue() Processing any unsent/unreceived MMS messages.
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service.c:activate_bearer() service 0xaaab
09610e80 setup 0 active 0
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service.c:activate_bearer() service 0xaaab
09610e80 waiting for 20 seconds
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:set_context() Setting C
ontext...
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service.c:mms_service_set_mmsc() service 0
xaaab09610e80 mmsc http://mms.invalid
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:set_context() Max numbe
r of bearers: 1
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:set_context() Attemptin
g to autoconfigure settings
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service-providers.c:mmsd_service_providers
_find_settings() mccmnc: 22256, apn: web
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:set_context() Current C
ontext APN: 'web', mmsd-tng settings MMS APN: 'apn.invalid'
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:set_context() The modem
 is not connected to the correct APN!
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:bearer_handler() Set MM
SC: 'http://mms.invalid', Set Proxy: '(null)', Set MMS APN: 'apn.invalid'
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:bearer_handler() At Bea
rer Handler: path '/org/freedesktop/ModemManager1/Modem/0' active '1' context_active '0'
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:bearer_handler() Error 
activating context!
Apr 24 19:40:45 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service.c:mms_service_bearer_notify() serv
ice=0xaaab09610e80 active=0 iface=(null) proxy=(null)
Apr 24 19:40:45 painfon mmsdtng[3942]: Could not find settings: Operator ID '22256' with APN 'web' 
not found in service provider database
Apr 24 19:40:45 painfon mmsdtng[3942]: Your MMS settings are not in the database! Please file a mer
ge request at https://gitlab.gnome.org/GNOME/mobile-broadband-provider-info so they can be added
Apr 25 11:07:48 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:cb_dbus_signal_sms_adde
[... 448 lines cut ...]
xaaab09610e80 mmsc http://mms.invalid
Apr 25 15:41:39 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:set_context() Max numbe
r of bearers: 1
Apr 25 15:41:39 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:set_context() Attemptin
g to autoconfigure settings
Apr 25 15:41:39 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service-providers.c:mmsd_service_providers
_find_settings() mccmnc: 22256, apn: web
Apr 25 15:41:39 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:set_context() Current C
ontext APN: 'web', mmsd-tng settings MMS APN: 'apn.invalid'
Apr 25 15:41:39 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:set_context() The modem
[... cut ...]
Apr 25 15:41:39 painfon mmsdtng[3942]: ../mmsd-2.6.2/src/service.c:mms_service_bearer_notify() serv
ice=0xaaab09610e80 active=0 iface=(null) proxy=(null)
Apr 25 15:41:39 painfon mmsdtng[3942]: Could not find settings: Operator ID '22256' with APN 'web' 
not found in service provider database
Apr 25 15:41:39 painfon mmsdtng[3942]: Your MMS settings are not in the database! Please file a mer
ge request at https://gitlab.gnome.org/GNOME/mobile-broadband-provider-info so they can be added
[alarm@painfon ~]$ ^C
[alarm@painfon ~]$ `

giorez, 2025-04-25:
I'm also trying to use last ASDP:
https://github.com/Biktorgj/quectel_eg25_recovery/blob/EG25GGBR07A08M2G_30.203.30.203/update/NON-HLOS.ubi

mikeshenson, 2025-05-16:
Your apn is not  set.
"Current C ontext APN: 'web', mmsd-tng settings MMS APN: 'apn.invalid' Apr 25 15:41:39 painfon mmsdtng[3942]: ../mmsd-2.6.2/plugins/modemmanager.c:set_context() The modem is not connected to the correct APN! "