the-modem-distro/pinephone_modem_sdk #250: Calls not working with simyo carrier.

Eloitor, 2025-04-02:
I tried versions 002, 003, 006 and 007.

I cannot receive calls (I get an SMS saying that I got a call while my phone was busy) nor send them.

I tried sending commands to +223344556677, but I didn't get any response (I remember this working some time ago)

mschilli87, 2025-04-02:
Same setup but different carrier works?

Eloitor, 2025-04-02:
I can't test it...

Eloitor, 2025-04-02:
I also tried https://github.com/Biktorgj/quectel_eg25_recovery but didn't work.

Eloitor, 2025-04-02:
I tried the command  `AT+QCFG=“ims”` and got `+QCFG: “ims”,0,0` I think this means the sim card does not support volte.

Eloitor, 2025-04-29:
I contacted simyo and they answered this (in spanish), but I didn't have time to try it. Let me know if you do.

[simyo.txt](https://github.com/user-attachments/files/19964563/simyo.txt)