the-modem-distro/pinephone_modem_sdk #253: No audio on EG25-G Minipcie/Usb Version

abdofallah, 2025-05-04:
I have tried both version v0.7.4 and v0.8.0 but when I try to play something on the soundcard, there is no audio recieved or played. I have preformed AT+EN_USBAUD.

gordon-quad, 2025-07-14:
Any luck with that?

I noticed that my UAC with stock firmware offers 8kHz sampling rate, but with SDK it is stuck with 16kHz, it might be relevant.

gordon-quad, 2025-07-14:
When I initiate call using `ATD` I see in logs unhandled messages from internal qmi but no call dispatch happens:
```
[105.9398] D init_internal_qmi_client: New QMI Message of 69 bytes
[105.9402] D dispatch_incoming_qmi_message: Pending message delivery service
[105.9404] I handle_incoming_voice_message: Start
[105.9405] I handle_incoming_voice_message: Unhandled message for the Voice Service: 002e
[105.9414] D init_internal_qmi_client: New QMI Message of 69 bytes
[105.9416] D dispatch_incoming_qmi_message: Pending message delivery service
[105.9417] I handle_incoming_voice_message: Start
[105.9418] I handle_incoming_voice_message: Unhandled message for the Voice Service: 002e
[106.4084] D start_scheduler_thread Tick!
[107.4091] D start_scheduler_thread Tick!
[107.5865] D init_internal_qmi_client: New QMI Message of 75 bytes
[107.5868] D dispatch_incoming_qmi_message: Pending message delivery service
[107.5871] I handle_incoming_voice_message: Start
[107.5872] I handle_incoming_voice_message: Unhandled message for the Voice Service: 002e
[107.5877] D init_internal_qmi_client: New QMI Message of 75 bytes
[107.5879] D dispatch_incoming_qmi_message: Pending message delivery service
[107.5882] I handle_incoming_voice_message: Start
[107.5883] I handle_incoming_voice_message: Unhandled message for the Voice Service: 002e
[108.2678] D init_internal_qmi_client: New QMI Message of 38 bytes
[108.2682] D dispatch_incoming_qmi_message: Pending message delivery service
[108.2684] I handle_incoming_voice_message: Start
[108.2685] I handle_incoming_voice_message: Unhandled message for the Voice Service: 0053
[108.2692] D init_internal_qmi_client: New QMI Message of 38 bytes
[108.2693] D dispatch_incoming_qmi_message: Pending message delivery service
```